orbingol/NURBS-Python #115: Evaluate Curves and Surfaces outside domain limits

rendetto, 2021-01-04:
This is more of a question, since I'm just an artist and the mathematics is not my field of expertise. 
Also I should apologize because I can only show images from a third party tool - namely Sverchok for Blender. 

Sverchok uses GEOMDL for some operations involving NURBS so that artists can benefit the advantages of curve/surface modeling approach. It also uses FreeCAD(OpenCascade) libraries in the same manner, so I used those for comparison.

The things in question I got are concerning the curves and surfaces definition outside the usual domain (please excuse me if I'm using the term domain incorrectly, but hopefully the pictures will be more explanatory)

I'll start with an example of a cubic curve using 4 control points and a periodic knot vector [0.0, 1.0, 2.0, 3.0, 4.0, 5.0, 6.0, 7.0].
The control points are set as being the corners of a unit square, so the setup is symmetric about the Y axis.

The picture I got looks like this:

[image]

The green spline is the generated GEOMDL curve, and the red one is the corresponding FreeCAD curve. Both curves use same control points and the same knot vector, but the results differ.

It seems both curves match in range [3.0, 7.0], but GEOMDL curve doesn't look intuitive for me in [0.0, 3.0] range, since it goes asymmetric regarding the Y axis.

Other effects appears when surfaces are evaluated. My findings are reported here https://github.com/nortikin/sverchok/issues/3521

The question I have is - are those considered bugs or an implementation specific behavior, because in my workflow this behavior tend to generate problems?

Don't hesitate to ask me for more info or examples on that matter. 

Thanks

orbingol, 2021-01-04:
@rendetto thanks for the question!

As a reference, I've created the following script according to your description in the question:

```
from geomdl import BSpline
from geomdl import utilities
from geomdl import exchange
from geomdl.visualization import VisMPL

# Create a B-Spline curve instance
curve = BSpline.Curve(normalize_kv=False)

# Set up the curve
curve.degree = 3
curve.ctrlpts = [[0.0, 0.0], [0.0, 1.0], [1.0, 1.0], [1.0, 0.0]]

# Auto-generate knot vector
curve.knotvector = [0.0, 1.0, 2.0, 3.0, 4.0, 5.0, 6.0, 7.0]
#curve.knotvector = [0.0, 0.0, 0.0, 0.0, 1.0, 1.0, 1.0, 1.0]

# Set evaluation delta
curve.delta = 0.01

# Evaluate curve
curve.evaluate()

# Plot the control point polygon and the evaluated curve
vis_comp = VisMPL.VisCurve2D()
curve.vis = vis_comp
curve.render()
```

It looks like it gets evaluated outside of the domain (which is [3.0, 4.0] for your case). I haven't tested outside of the domain NURBS evaluation. There might be an issue with it. I'll look into it and get back to you.

orbingol, 2021-01-14:
It seems like there was a wrong assumption in the span finding function which only affects out of domain evaluation. After the fix, it seems to be computing the curve points correctly.

[image]

I'll push the changes to `5.x-dev` branch after some tests.

rendetto, 2021-01-14:
Awesome! Thank you very much! 

There are kind of similar problems with the surfaces. I'll try to create and show you some simple test cases. 
Did you prefer me to open another issue or I can upload the examples here?

orbingol, 2021-01-14:
It would be great if you could test the surfaces with the `5.x-dev` branch. Because surface and curves use the same span finding and basis computation functions. Theoretically, a fix for curves should also fix the surfaces. I'd be happy if would report any issues under this ticket.

rendetto, 2021-01-14:
I just tested both the curves and the surfaces with the latest 5.x-dev branch. The issues I got are gone.

Thanks so much!

orbingol, 2021-01-23:
Fixed with v5.3.1 release.

closed 2021-01-23 by orbingol