orbingol/NURBS-Python #116: VisPlotly and Jupyter

ediemiglio, 2021-01-12:
Hi Onur, thanks for the great library! I have a question: when I am using plotly inside Jupyter (or Colab) the resulting figure is shown inside the notebook. On the other hand when I am using VisPlotly for visualizing a NURBS surface the result is an html file (as described in the documentation of NURBS-Python). I was wondering if there is any method to obtain the figure in the notebook. 
Thanks in advance.

orbingol, 2021-01-14:
@ediemiglio thanks for the request!

I thought Jupyter support was there but I am guessing I removed it due to some incompatibility issues a while ago (they changed their API a couple of times since I implemented `VisPlotly`). I'll try to add it back for the updated Plotly API.

orbingol, 2021-01-14:
I've added the Jupyter and Plotly renderer support in `5.x-dev` branch. Could you test it @ediemiglio?

The default behavior of `VisPlotly` hasn't changed, it will save the figure as an HTML file.

`VisPlotly` class in `5.x-dev` branch will show the following in the Jupyter notebook automatically:

```
from geomdl import BSpline
from geomdl import utilities
from geomdl.visualization import VisPlotly as vis

curve = BSpline.Curve()
curve.degree = 3
curve.ctrlpts = [[10, 5, 10], [10, 20, -30], [40, 10, 25], [-10, 5, 0]]
curve.knotvector = utilities.generate_knot_vector(curve.degree, len(curve.ctrlpts))

curve.vis = vis.VisCurve3D()
curve.render()
```

You will be able to change the renderer. For instance:

```
from geomdl import BSpline
from geomdl import utilities
from geomdl.visualization import VisPlotly as vis

# Set Plotly default renderer
from plotly import io as pio
pio.renderers.default = "pdf"  # pio.renderers() will show the available renderers

curve = BSpline.Curve()
curve.degree = 3
curve.ctrlpts = [[10, 5, 10], [10, 20, -30], [40, 10, 25], [-10, 5, 0]]
curve.knotvector = utilities.generate_knot_vector(curve.degree, len(curve.ctrlpts))

# Pass "use_renderer" keyword argument to the configuration for using the default renderer
curve.vis = vis.VisCurve3D(use_renderer=True)
curve.render()
```

ediemiglio, 2021-01-14:
Thanks. I will try the dev branch.
   Edie

orbingol, 2021-01-23:
I am closing this issue. It is fixed with v5.3.1 release.

closed 2021-01-23 by orbingol