orbingol/NURBS-Python #118: Little discrepancy

sonomirco, 2021-01-20:
**Describe the bug**
I have done some tests with NurbsPython and VerbNurbs and I got the same results.
But when I compared those results with Rhino, for instance, I got a little discrepancy and I would like to know what can be the cause?

var knots = { 0.0, 0.0, 0.0, 0.0, 0.33, 0.66, 1.0, 1.0, 1.0, 1.0 };
var degree = 3;
var controlPts = {{5,5,0},{10, 10, 0},{20, 15, 0},{35, 15, 0},{45, 10, 0},{50, 5, 0}};

these are the results from NurbsPython and VerbNurbs I got using the parameter 0.5 to define a point on the curve and the tangent.

pt -> { 27.64549230676716, 14.691702224146088, 0 }
tanget -> { 0.9999839615943609, -0.005663616693228194, 0 }

these are the result from Rhino.
pt -> {27.5, 14.6875, 0.0}
tanget -> {1.0, 0.0, 0.0}

Do you think is a tolerance issue? Or can it be due to the use of float and double values?

orbingol, 2021-01-21:
@sonomirco thanks for the question! I don't think it is a bug or tolerance, but it looks like a precision issue. Since the integrity of geomdl, verb and Rhino is in question, it might be a good idea to use another software for validation (or we would be solving everything on the paper). My preference is GNU Octave.

Here is a GNU Octave script with the values you provided (I used v6.1.0 x64 with nurbs v1.13.3 module):

```
# Housekeeping
clear all;

# Load "nurbs" package
pkg load nurbs

# Control points and knot vector for a cubic curve
ctrlpts = [05 10 20 35 45 50; 05 10 15 15 10 05; 0 0 0 0 0 0];
kv = [0.0, 0.0, 0.0, 0.0, 0.33, 0.66, 1.0, 1.0, 1.0, 1.0];

# Generate the NURBS curve
crv = nrbmak(ctrlpts, kv);

# Evaluate the curve point at u = 0.5
[p, w] = nrbeval(crv, 0.5);

# Compute the 1st and 2nd derivative curves
[ders1, ders2] = nrbderiv(crv);

# Evaluate the derivatives
[p1, dp1] = nrbdeval(crv, ders1, 0.5);

# Normalize derivative vectors
dp1n = vecnorm(dp1);

# Results
# p = [27.465, 14.692, 0]
# dp1n = [0.9998, -0.0056636, 0]
```

where `p` is the curve point and `dp1n` is the tangent vector.

Let's update the knot vector like this: `kv = [0.0, 0.0, 0.0, 0.0, 0.333333, 0.666667, 1.0, 1.0, 1.0, 1.0];`

And the results will be:

```
# Results
# p = [27.5, 14.687, 0]
# dp1n = [1, 0, 0]
```

I hope it is more clear now.

sonomirco, 2021-01-21:
Hi @orbingol,
thanks for the replay, I knew it wasn't a bug but I didn't found a way to change the label.
Also, a couple of hrs before your message, I found the answer exactly on the precision of the knots!
Good idea to use Octave!!!
Thanks again and NurbsPython is a very nice library!

closed 2021-01-21 by sonomirco