orbingol/NURBS-Python #125: Degree elevation modifies NURBS curve shape

lslore, 2021-03-14:
**Describe the bug**
Degree elevation operation via **operations.degree_operations** modifies the shape of the NURBS curve for particular input curves. 

The problem seems to arise when calling **split_curve** and **link_curves** in **operations.degree_operations**. The target NURBS curve is a near 90deg arc and the splitting operation returns the arc itself and an infinitesimal segment, which generates strange behaviour when linked. 

**To Reproduce**
Following code reproduces the behaviour:

```
from geomdl import NURBS, operations
from geomdl.visualization import VisMPL

# Define curve
curve = NURBS.Curve()
curve.degree = 2
curve.ctrlpts = [[821.51962938, -677.395551576, 3.5410634089],
 [873.5806602364927, -682.3571637856359, 88.77621736805291],
 [879.1302220964927, -582.5406209926358, 91.19699162172292],
 [879.1656602459713, -581.9032177447318, 91.21245004565435],
 [879.196849813, -581.265461848, 91.2209560929]]
curve.knotvector = [0.0, 0.0, 0.0, 0.5, 0.5, 1.0, 1.0, 1.0]
curve.weights = [1.0, 0.7071067811865476, 1.0, 0.9999796117372515, 1.0]

curve.vis = VisMPL.VisCurve3D()
curve.render()

# Elevate curve degree
degree_elevated = operations.degree_operations(curve, [1])
degree_elevated.vis = VisMPL.VisCurve3D()
degree_elevated.render()

```
**Expected Behavior**
Degree elevation operation should not modifz the shape of the input curve. 

**Configuration:**
 - OS: Windows 10
 - Python distribution: Anaconda
 - Python version: Python 3.6.10
 - geomdl install source:  Anaconda
 - geomdl version/branch: 5.3.1, py_0

**Screenshots (Optional)**
[image]

mcmestriner, 2024-09-13:
Hello from the future (September 2024).

I guess the problem is the number of knot removals in function degree_operations (operations.py)

original code: `num = obj.degree + 1`
should be: `num = obj.degree`

GarthSnyder, 2025-05-21:
May be related to #135

closed 2025-05-23 by orbingol