orbingol/NURBS-Python #130: How to increase number of triangles in .stl file?

sphh, 2021-06-02:
I created an aerofoil with the help of this wonderful module (Thank you!).

Now I want to export it as a .stl file. That works well, but the tesselation is quite coarse. Because in #128 it is mentioned, that increasing/decreasing the surfaces' `delta` values would change the resolution, I divided the default delta value by 100. Unfortunately that does not change the number of triangles.

Then I played with the keyword argument `vertex_spacing` of the `export_stl()` function. If I set it to a value >1, I get a coarser grid. If I set it to 0.1, I get a `geomdl.exceptions.GeomdlException: GEOMDL ERROR: Vertex spacing should be bigger than zero` error.

What is the proper way to refine the grid (increase the number of triangles), so that the exported file follows the surface better? Thanks!!

orbingol, 2021-06-03:
The `delta` value needs to change the number of triangles as it changes the number of evaluated points. `delta` property is stored as a tuple (as you may have already noticed), so dividing a tuple by an integer might not work correctly.

There are also dimensional counterparts of `delta`, like `delta_u`, `delta_v`, and `delta_w` which can be used to update the number of evaluated points on the specified parametric direction.

You could also play with the `sample_size` property. It is a wrapper for the `delta` value but makes more sense to some users.

You could also call `reset` and then `evaluate` methods to make sure that the delta/sample size values are updated (this step shouldn't be necessary but let's make sure of it).

sphh, 2021-06-03:
I found the mistake: I set the `delta` value for my surfaces and then I added these surfaces to a `SurfaceContainer` assuming that the final tessellation of the container takes the `delta` value for each surface. That was wrong: I have to set the `delta` value for the `SurfaceContainer`!

PS: By dividing the `delta` value by 100 I obviously meant to do it element-wise.

closed 2021-06-03 by sphh