orbingol/NURBS-Python #133: STEP export

sphh, 2021-06-24:
I read issue #70 and just wanted to ask, if you could make any progress in exporting STEP files?

Currently I use this toolchain:
`geomdl > .json` → `json2on > .3dm` → `FreeCAD > .step`

It would be really great, if it would just be `geomdl > .step`, especially because the FreeCAD step involves a GUI (mind you, it could be scripted, but that would require me digging into the FreeCAD API, but on the positive side – with even more digging! – it might also be possible to read the .json file directly into FreeCAD.).

I would very much prefer to avoid chained conversions and have the STEP file written directly from `geomdl`. What are the chances to get one?

I never thanked you for this – your – module. It is so useful and easy to use!

orbingol, 2021-06-25:
@sphh thanks for your kind words. I was working on the STEP export feature before the pandemic. It wasn't going fast, to be honest. I was planning to use OpenCASCADE and create some apps very similar to `json2on` for exporting to IGES and STEP.

I still have the plan to develop the exporter apps but it is very hard to provide a schedule right now. I am hoping to change that very soon but it will definitely take time to develop and test the apps.

The good thing is that there are some packages that provide python bindings for OpenCASCADE. I am guessing FreeCAD is also using OpenCASCADE but I am not sure if FreeCAD provides an API or not.

* https://github.com/tpaviot/pythonocc-core -- we used this in a project some time ago and we liked it
* https://github.com/trelau/pyOCCT -- just found it

OpenCASCADE is relatively easy to use compared to the others that I used. The package maintainers use SWIG and pybind11 so it should be straightforward to port C++ examples to Python.

portnov, 2021-06-25:
Another possibility you may want to explore is Python API of FreeCAD. If you install FreeCAD and add it's libraries directory (with FreeCAD.dll / FreeCAD.so) to python's sys.path, `import FreeCAD` becomes available. It allows to export Step, Iges, Brep files among other things.

sphh, 2021-06-27:
@orbingol, you are absolutely right: According to https://wiki.freecadweb.org/OpenCASCADE, FreeCAD uses OpenCASCADE.

Am I right, that for a conversion to all the file formats supported by OpenCASCADE, that library can be used instead of the full-fledged FreeCAD installation?

orbingol, 2021-07-03:
> Am I right, that for a conversion to all the file formats supported by OpenCASCADE, that library can be used instead of the full-fledged FreeCAD installation?

I haven't investigated in detail but I think so.

I am closing this issue for now. Feel free to reopen if you have questions.

closed 2021-07-03 by orbingol