orbingol/NURBS-Python #144: Conda package

germa89, 2022-02-03:
I'm working in a project that one of the requirements is this package. We want to put our package is conda, but it seems it is easier to do this if all the packages dependencies are in conda already. 

So I'm happy to do the recipe if any of you agrees to be the maintainer (mainly you will be cited in their page as maintainer)

orbingol, 2022-02-04:
@germa89 already there: https://anaconda.org/orbingol/geomdl

The recipes are under `conda` directory of this repo.

closed 2022-02-04 by orbingol

germa89, 2022-02-07:
@orbingol would it be possible to have in conda-forge too? 

https://conda-forge.org/
https://github.com/conda-forge/staged-recipes

orbingol, 2022-02-08:
Sure, feel free to submit the package request to conda-forge.

germa89, 2022-02-08:
Thank you! I will do it shortly!