orbingol/NURBS-Python #165: Visualization (VisMPL) is showing an empty figure

curiouscivilengineer, 2023-04-14:
**Describe the bug**
I want to create NURBS Surfaces with geomdl. I installed the package via conda and started by trying to run the example given here : https://nurbs-python.readthedocs.io/en/5.x/visualization.html 

This is the given code:
`from geomdl import BSpline
from geomdl.visualization import VisMPL

# Control points
ctrlpts = [
    [[-25.0, -25.0, -10.0], [-25.0, -15.0, -5.0], [-25.0, -5.0, 0.0], [-25.0, 5.0, 0.0], [-25.0, 15.0, -5.0], [-25.0, 25.0, -10.0]],
    [[-15.0, -25.0, -8.0], [-15.0, -15.0, -4.0], [-15.0, -5.0, -4.0], [-15.0, 5.0, -4.0], [-15.0, 15.0, -4.0], [-15.0, 25.0, -8.0]],
    [[-5.0, -25.0, -5.0], [-5.0, -15.0, -3.0], [-5.0, -5.0, -8.0], [-5.0, 5.0, -8.0], [-5.0, 15.0, -3.0], [-5.0, 25.0, -5.0]],
    [[5.0, -25.0, -3.0], [5.0, -15.0, -2.0], [5.0, -5.0, -8.0], [5.0, 5.0, -8.0], [5.0, 15.0, -2.0], [5.0, 25.0, -3.0]],
    [[15.0, -25.0, -8.0], [15.0, -15.0, -4.0], [15.0, -5.0, -4.0], [15.0, 5.0, -4.0], [15.0, 15.0, -4.0], [15.0, 25.0, -8.0]],
    [[25.0, -25.0, -10.0], [25.0, -15.0, -5.0], [25.0, -5.0, 2.0], [25.0, 5.0, 2.0], [25.0, 15.0, -5.0], [25.0, 25.0, -10.0]]
]

# Create a BSpline surface
surf = BSpline.Surface()

# Set degrees
surf.degree_u = 3
surf.degree_v = 3

# Set control points
surf.ctrlpts2d = ctrlpts

# Set knot vectors
surf.knotvector_u = [0.0, 0.0, 0.0, 0.0, 1.0, 2.0, 3.0, 3.0, 3.0, 3.0]
surf.knotvector_v = [0.0, 0.0, 0.0, 0.0, 1.0, 2.0, 3.0, 3.0, 3.0, 3.0]

# Set evaluation delta
surf.delta = 0.025

# Evaluate surface points
surf.evaluate()

# Import and use Matplotlib's colormaps
from matplotlib import cm

# Plot the control points grid and the evaluated surface
surf.vis = VisMPL.VisSurface()
surf.render(colormap=cm.cool)`

The code runs with without an error. 

**Expected Behavior**
It should look like this:
[image]
But it just looks like this:
[image]

**Configuration:**
 - OS: [Windows 11]
 - Python distribution: [Anaconda]
 - Python version: 3.10.10
 - geomdl install source: Anaconda
 - geomdl version/branch: 5.3.1
 - I also tried deleting and reinstalling both anaconda and geomdl, without succes

It would be really appreciate your help.

jcs15c, 2023-05-23:
I was able to fix it, temporarily at least, by replacing line 431 in `visMPL.py`,

```python
ax = Axes3D(fig)
```

to 

```python
ax = fig.add_subplot( 111, projection='3d' )
```

dalbenzioG, 2023-05-26:
Thank you! Same problem and this also solve my issue :)

Siltyx, 2023-06-02:
Same and solved. Also, for anyone having the error that `np.float` is deprecated, just replace with the inbuilt `float` type as NumPy prompts and you're good.

Lan998, 2023-10-31:
ax=Axes3D(fig);fig.add_axes(ax)

Tomoki-0526, 2023-12-06:
I've tried the 2 methods mentioned above but they didn't work. Anyone has another way?

closed 2025-05-26 by orbingol