orbingol/NURBS-Python #171: Problem in Visualization for Sample code

newton-raphson, 2023-11-24:
**Describe the bug**
`np.float` was a deprecated alias for the builtin `float`. To avoid this error in existing code, use `float` by itself. Doing this will not modify any behavior and is safe. If you specifically wanted the numpy scalar type, use `np.float64` here.

**To Reproduce**
Steps to reproduce the behavior:
1. pip install geomdl
2. run the sample code to produce the code 
3. AttributeError: module 'numpy' has no attribute 'float'.
`np.float` was a deprecated alias for the builtin `float`. To avoid this error in existing code, use `float` by itself. Doing this will not modify any behavior and is safe. If you specifically wanted the numpy scalar type, use `np.float64` here.
The aliases was originally deprecated in NumPy 1.20; for more details and guidance see the original release note at:

**Expected Behavior**

**Configuration:**
 - macos M1
 - python3.11
 - pip - geomdl version/branch:

Srallars97, 2024-02-14:
Got the same problem.

I fixed by changing l.71 in geomdl/visualization/VisMPL.py.

self.dtype = np.float

to 
self.dtype = float
.

Solved it for me.

TheOtherRealm, 2024-06-22:
Same issue, the full stack trace from running the plotting example on the "Basics" page in the documentation gives:
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
File c:\Develop\things_to_print\headset.py:34
     31 mcrv = multi.CurveContainer(crv1, crv2)
     33 # Import Matplotlib visualization module
---> 34 from geomdl.visualization import VisMPL
     36 # Set the visualization component of the curve container
     37 mcrv.vis = VisMPL.VisCurve3D()

File c:\Users\someone\micromamba\envs\things_to_print\Lib\site-packages\geomdl\visualization\VisMPL.py:124
    116         """ Detects if Jupyter notebook GUI toolkit is active
    117 
    118         return: True if the module is running inside a Jupyter notebook
    119         rtype: bool
    120         """
    121         return True if "nbAgg" == mpl.get_backend() else False
--> 124 class VisCurve2D(vis.VisAbstract):
    125     """ Matplotlib visualization module for 2D curves """
    126     def __init__(self, config=VisConfig(), **kwargs):

File c:\Users\someone\micromamba\envs\things_to_print\Lib\site-packages\geomdl\visualization\VisMPL.py:126, in VisCurve2D()
    124 class VisCurve2D(vis.VisAbstract):
    125     """ Matplotlib visualization module for 2D curves """
--> 126     def __init__(self, config=VisConfig(), **kwargs):
    127         super(VisCurve2D, self).__init__(config, **kwargs)
    129     def render(self, **kwargs):

File c:\Users\someone\micromamba\envs\things_to_print\Lib\site-packages\geomdl\visualization\VisMPL.py:71, in VisConfig.__init__(self, **kwargs)
     69 def __init__(self, **kwargs):
     70     super(VisConfig, self).__init__(**kwargs)
---> 71     self.dtype = np.float
     72     self.display_ctrlpts = kwargs.get('ctrlpts', True)
     73     self.display_evalpts = kwargs.get('evalpts', True)

File c:\Users\someone\micromamba\envs\things_to_print\Lib\site-packages\numpy\__init__.py:324, in __getattr__(attr)
    319     warnings.warn(
    320         f"In the future `np.{attr}` will be defined as the "
    321         "corresponding NumPy scalar.", FutureWarning, stacklevel=2)
    323 if attr in __former_attrs__:
--> 324     raise AttributeError(__former_attrs__[attr])
    326 if attr == 'testing':
    327     import numpy.testing as testing

AttributeError: module 'numpy' has no attribute 'float'.
`np.float` was a deprecated alias for the builtin `float`. To avoid this error in existing code, use `float` by itself. Doing this will not modify any behavior and is safe. If you specifically wanted the numpy scalar type, use `np.float64` here.
The aliases was originally deprecated in NumPy 1.20; for more details and guidance see the original release note at:
    https://numpy.org/devdocs/release/1.20.0-notes.html#deprecations